waterloo-rocketry/cansw_canards #1: Understand how to use the new Polulu IMU

celery6, 2024-10-11:
Goal is to become familiar with this IMU and it's quirks. 

- find and read the IMU datasheet (pls upload to Drive somewhere), understand which registers can/need to get initialized
- get wires, a nucleo devkit, hook it up, initialize registers if necessary, try to read sensor values. Prob easiest to create a new stm32cubeide project for the nucleo for this testing? not sure 
- see[ last year's ICM handler ](https://github.com/waterloo-rocketry/cansw_processor_airbrakes/blob/master/STM32Cube/Tasks/ICM-20948.c)for example; copy paste as necessary to get a simple bare-metal dev setup going for this 
- write info down somewhere in the drive

This isn't formal production code or anything, just temporary dev setup to let us read the imu asap. And to understand what registers need to be initialized and how.

Ex. a few years ago, minisensor board's ICM registers were incorrectly configured so it never actually read gyro values 💀 Ideally we take our time to understand how to use the IMU correctly!

arpitarunkumaar, 2024-10-16:
dump info here: https://docs.google.com/document/d/1PSHZkS6BaPKJO4hi8oGO6AXEStlA4ESh5UQzXhF-fkc/edit?usp=sharing

celery6, 2025-02-20:
good enough