waterloo-rocketry/cansw_canards #2: Determine task priorities

celery6, 2024-10-13:
We need to learn more about how to determine task priorities so it's not so arbitrary! This requires also understanding the interactions between tasks, particularly which is blocking each other and stuff like that

celery6, 2024-12-22:
adding all build output to this - total program size, task stack sizes, etcetc will think of more probably

Joe-Joe-Joe-Joe, 2025-02-20:
Its done in config not in source code, but make sure we configure
- `TIMER_TASK_PRIORITY`
- `TIMER_QUEUE_LENGTH`
- `TIMER_TASK_STACK_DEPTH`

Also we should optimize heap size if possible. Although that might introduce more problems than it solves

celery6, 2025-05-05:
timer configured correctly, all our tasks configured logically + confirmed in testing. release build is alive too

closed 2025-05-05 by celery6