waterloo-rocketry/cansw_canards #3: How to do unit testing

celery6, 2024-10-13:
How??? Should pick an existing framework, make sure it's easy for devs to pick up smoothly. And integrate into existing dev environment (stm32 world)

I have no practical experience in this.

Joe-Joe-Joe-Joe, 2024-10-16:
disclaimer: I know absolutely nothing about using frameworks intended for C++ for C development. I do not remember if cubeIDE natively compiles everything with g++ regardless. If it does maybe you don't need to worry about compatibility so much.

Anyway this one looks good: https://github.com/doctest/doctest
Any of the ones you get with a google search are probably good enough as a first step. Main thing is to pick one and run with it. (And not use Zephyr lol)

celery6, 2024-11-21:
decision: use googletest because its one of the most popular and well-documented. It is cpp, but theres many recommendations for using cpp test framework to test c code so this is not an issue. Also likely https://github.com/meekrosoft/fff for implementing test mocks or a better one if we find one, but this seems to be commonly paired with gtest.

Resulting workflow might look like:
- edit the project in stm32cubeide as usual
- separate script for building unit tests (not through cubeide):
  - only build the files in `src/` using gcc (ie, build for an executable binary, not for flashing onto an stm32). Avoid building the HAL or freertos?
  - build the tests using gcc too, and link the processor build as a library
  - run the test binary locally

Things to figure out:
- build system: is it worth trying to integrate unit testing into stm32cubeide (aka eclipse CDT) or should we just use cmake? It doesn't seem like the ide would provide anything useful for unit testing anyway, and just make it harder to make the build system.
    - seems recommended to compile source code separately from the test code, using C and cpp compilers respectively. is that kind of config even possible in eclipse cdt???
- Mocking HAL and freertos calls? what should be mocked? what can we keep? How do you include mocks in the build system?

maybe helpful: [this book](https://ocul-wtl.primo.exlibrisgroup.com/permalink/01OCUL_WTL/vk29fk/alma999986595612605162) which is like the bible of embedded C testing. And its corresponding code which has a setup example https://github.com/jwgrenning/tddec-code (not googletest, but similarly cpp framework + c source code)

riyapatel25, 2024-11-25:
@celery6 I know Orbital uses gtest as well and  compiles unit tests seperately than main firmware, don't think integrating to ide would be worth it. This might be helpful: https://github.com/UWOrbital/OBC-firmware/tree/main/test

closed 2025-01-16 by celery6