waterloo-rocketry/cansw_canards #22: i2c module design

celery6, 2024-10-18:
see #5 

Provide an interface for reading/writing i2c. Essentially wrap the stm32 HAL calls and configuration, but also consider how to make this rtos-friendly for our purposes. Are calls blocking? Using interrupts? ???

Things to figure out:
- There was some discussion about rolling i2c into the IMU device drivers instead of having it also a standalone module. Look at the stm32 hal i2c functions and see if there’s any setup required that would warrant its own module, or if it’s convenient to just call hal directly from the IMU module. Last year had the “my2c” driver (i2c but my2c) which setup mutexes. Was this necessary? Can we avoid that and still make it work with rtos?

Joe-Joe-Joe-Joe, 2024-10-18:
- If you want maximum speed use multiple I2C peripherals, one for each device
- Otherwise, I think you ought to put in a gatekeeper task. Same way we did CAN last year. See https://electronics.stackexchange.com/questions/368015/multi-thread-i2c-synchronization

Joe-Joe-Joe-Joe, 2024-10-18:
I would try to design for the case where you have 2 state estimation threads running in parallel (as seems to be done in some redundant configurations I was looking at), one for each IMU

Joe-Joe-Joe-Joe, 2024-10-18:
They say in that post I linked, give it a pointer to a data buffer (presumably for the read data), but this obviously begs the question of "how does the calling task know that the buffer it passed has been filled". I suppose you could ship a semaphore along with it?
so you would have like
`struct 
{
uint32_t cmd_len
uint8_t I2C_addr
struct I2Ctransaction *commands
uint32_t data_len
uint8_t *data
SemaphoreHandle_t done
}`

Joe-Joe-Joe-Joe, 2024-10-19:
[image]
eh?

Joe-Joe-Joe-Joe, 2024-10-24:
https://docs.google.com/document/d/12RiXJ1J8ImkTuui0jkFiIdUy7xML1SvxLy82ACC8qKg/edit?tab=t.0#heading=h.fmdwnyb5c0ih

celery6, 2024-11-09:
update: board respin puts only 1 i2c device on each i2c bus, and each device has its own handler thread so there will be no multithreaded i2c conflicts. task scope decrease!!

closed 2025-01-12 by taisirhassan