waterloo-rocketry/cansw_canards #25: implement a static code analyzer into our workflow

celery6, 2024-10-21:
the rocketry embedded standard (still wip) decided that each project is responsible for implementing their own coding safety standards (or not implementing).

since processor board has potential to be safety-critical especially in coming years, we need to start taking steps to enforce a safety standard.

top choices rn are:
Barr C standard https://barrgroup.com/embedded-systems/books/embedded-c-coding-standard
JPL C standard
CERT Secure C (maybe? this seems more for security, like hackers than embedded avionics)

choosing between these (or any other one not listed) mostly depends on how doable it is to enforce these via static analyzers. There are many commercial static analysis tools but we aren't paying for those. 

This task is to find a free static analyzer that can enforces one of these C safety standards, and can be integrated into our workflow (stm32cubeide) conveniently.

Joe-Joe-Joe-Joe, 2024-10-29:
@JasonBrave I don't know the first thing about clang please help
How do i tell it that I want it to go over an entire directory so it knows that includes are there?
In cubeIDE you specify the include directories and some stuff happens under the hood with Make that is obfuscated from the end user. 
[image]

Joe-Joe-Joe-Joe, 2024-12-22:
Now that we are separating out our workflow into a platform-agnostic build (rather than trying to integrate with cubeIDE) in #43 , this should be much easier, right?

celery6, 2025-01-12:
cppcheck seems to be the most reasonable static analyzer. can't find any other popular opensource and easy to integrate. ill make a branch for this asap

celery6, 2025-04-09:
cppcheck deemed not very helpful as it catches practically 0 issues that -werror and formatting wont already catch. for reference, running on entire airbrakes codebase produced 0 warnings ?? maybe im usin it wrong.. or we literally dont use complex C features that would raise errors

closed 2025-04-09 by celery6