waterloo-rocketry/cansw_canards #29: investigate upgrading freertos version

celery6, 2024-11-13:
yup

martendo, 2024-11-13:
The story so far:
- The current files in [STM32Cube/Middlewares/Third_Party/FreeRTOS/Source/](https://github.com/waterloo-rocketry/cansw_processor_canards/tree/20d180ac136e64270102cefd403d4c72e3df854c/STM32Cube/Middlewares/Third_Party/FreeRTOS/Source) suggest that we are using FreeRTOS V10.3.1, released 18 February 2020.
  - This appears to be the latest version provided by STM32CubeIDE &mdash; other people mentioning this: [exhibit A](https://community.st.com/t5/stm32cubemx-mcus/how-to-update-freertos-from-v10-3-1-to-the-latest-version-v10-4/td-p/64623), [exhibit B](https://community.st.com/t5/stm32cubeide-mcus/upgrading-freertos-in-cubemxide/td-p/630463).
- The latest version as of today is V11.1.0, released 22 April 2024. Many changes have been made since V10.3.1: [FreeRTOS release history](https://w.freertos.org/Documentation/04-Roadmap-and-release-note/02-Release-notes/00-Release-history).
  - One change we are potentially interested in is the addition of support for multiple task notification values per task (would be nice for the logger), added in V10.4.0.
- We can't update FreeRTOS through STM32CubeIDE, but it appears that the source files for FreeRTOS can be installed into the project manually: [someone's instructions](https://github.com/AnielShri/STM32_FreeRTOS_Manual_Addition).
  - The configuration in our IOC conveniently matches the settings given in the above instructions! I would still be wary of trusting all of it to be correct though.
  - Testing a replacement of FreeRTOS source files with V11.1.0 was successful with very minor changes:
    - ```diff
      -void vApplicationStackOverflowHook (TaskHandle_t xTask, signed char *pcTaskName)
      +void vApplicationStackOverflowHook (TaskHandle_t xTask, char *pcTaskName)
      ```
    - Discrepancy between new (?) `INCLUDE_xTaskDelayUntil` and old (?) `INCLUDE_vTaskDelayUntil`
    ```diff

celery6, 2024-11-13:
my own concern abt upgrading is possibly not as stable as LTS versions. but this is freertos and the version isnt that new so it should be fine prob

celery6, 2024-11-17:
> Discrepancy between new (?) INCLUDE_xTaskDelayUntil and old (?) INCLUDE_vTaskDelayUntil: 

We should be migrating to `xTaskDelayUntil` instead of the old `vTaskDelayUntil` so this makes sense

celery6, 2025-01-11:
descoping this until next design cycle :pensive:

closed 2025-03-11 by Joe-Joe-Joe-Joe